KurtE/ILI9341_t3n #4: Async Compiling with latest Teensy Loader and Arduino IDE

Donziboy2, 2017-10-01:
Managed to get the Async branch to compile with the Latest Teensy Loader 1.39, Arduino IDE and SPIN that comes with 1.39.
Compiler kept getting stuck on the 2 below calls in a few places.
Not idea if it changes anything still trying to get dma working.

	_pkinetisk_spi = _pspin->kinetisk_spi();
	_pkinetisl_spi = _pspin->kinetisl_spi();

`// https://github.com/PaulStoffregen/ILI9341_t3n
// http://forum.pjrc.com/threads/26305-Highly-optimized-ILI9341-(320x240-TFT-color-display)-library

/***************************************************
  This is our library for the Adafruit ILI9341 Breakout and Shield
  ----> http://www.adafruit.com/products/1651

  Check out the links above for our tutorials and wiring diagrams
  These displays use SPI to communicate, 4 or 5 pins are required to
  interface (RST is optional)
  Adafruit invests time and resources providing this open source code,
  please support Adafruit and open-source hardware by purchasing
  products from Adafruit!

  Written by Limor Fried/Ladyada for Adafruit Industries.
  MIT license, all text above must be included in any redistribution
 ****************************************************/
// <SoftEgg>

//Additional graphics routines by Tim Trzepacz, SoftEgg LLC added December 2015
//(And then accidentally deleted and rewritten March 2016. Oops!)
//Gradient support
//----------------
//		fillRectVGradient	- fills area with vertical gradient
//		fillRectHGradient	- fills area with horizontal gradient
//		fillScreenVGradient - fills screen with vertical gradient
// 	fillScreenHGradient - fills screen with horizontal gradient

//Additional Color Support
//------------------------
//		color565toRGB		- converts 565 format 16 bit color to RGB
//		color565toRGB14		- converts 16 bit 565 format color to 14 bit RGB (2 bits clear for math and sign)
//		RGB14tocolor565		- converts 14 bit RGB back to 16 bit 565 format color

//Low Memory Bitmap Support
//-------------------------
// 		writeRect8BPP - 	write 8 bit per pixel paletted bitmap
// 		writeRect4BPP - 	write 4 bit per pixel paletted bitmap
// 		writeRect2BPP - 	write 2 bit per pixel paletted bitmap
// 		writeRect1BPP - 	write 1 bit per pixel paletted bitmap

//TODO: transparent bitmap writing routines for sprites

//String Pixel Length support
//---------------------------
//		strPixelLen			- gets pixel length of given ASCII string

// <\SoftEgg>

#include "ILI9341_t3n.h"
#include <SPIN.h>
#include <SPI.h>  

//#define DEBUG_ASYNC_UPDATE  // Enable to print out dma info

#ifdef ENABLE_ILI9341_FRAMEBUFFER
DMASetting 	ILI9341_t3n::_dmasettings[4];
DMAChannel 	ILI9341_t3n::_dmatx;

DMAChannel 	_dmatx;
ILI9341_t3n *ILI9341_t3n::_dmaActiveDisplay = 0;
volatile uint8_t  	ILI9341_t3n::_dma_state = 0;  // Use pointer to this as a way to get back to object...
volatile uint32_t	ILI9341_t3n::_dma_frame_count = 0;	// Can return a frame count...
[... 2689 lines cut ...]
	if (! inverted) {
		fill = _fillcolor;
		outline = _outlinecolor;
		text = _textcolor;
	} else {
		fill =  _textcolor;
		outline = _outlinecolor;
		text = _fillcolor;
	}
	_gfx->fillRoundRect(_x - (_w/2), _y - (_h/2), _w, _h, min(_w,_h)/4, fill);
	_gfx->drawRoundRect(_x - (_w/2), _y - (_h/2), _w, _h, min(_w,_h)/4, outline);
	_gfx->setCursor(_x - strlen(_label)*3*_textsize, _y-4*_textsize);
	_gfx->setTextColor(text);
	_gfx->setTextSize(_textsize);
	_gfx->print(_label);
}

bool Adafruit_GFX_Button::contains(int16_t x, int16_t y)
{
	if ((x < (_x - _w/2)) || (x > (_x + _w/2))) return false;
	if ((y < (_y - _h/2)) || (y > (_y + _h/2))) return false;
	return true;
}

`

KurtE, 2017-12-02:
I updated the master branch to use the newer SPIN method call.

The function to get the  kinetisk_spi object changed names.

Donziboy2, 2017-12-04:
I should be able to test it in the next week or 2, tackling my 18bit ADC and how to work it into code without getting in the way of other things.

Donziboy2, 2017-12-05:
Most of my saved examples use Async and framebuffer so they fall flat on their faces.  The oldest test code I have that does not use Async also fails.

Here is the error I get and the test file I used.

[Error.txt](https://github.com/KurtE/ILI9341_t3n/files/1529249/Error.txt)

[ILI9341 Test.txt](https://github.com/KurtE/ILI9341_t3n/files/1529248/ILI9341.Test.txt)

KurtE, 2020-03-18:
Closing now as it is over 2 years old 

Let me know if there is something that still should be looked at.

closed 2020-03-18 by KurtE