KurtE/ILI9341_t3n #20: Teensy 3.6 vs Teensy 4.0 DMA Speed and frameCount()

mr-stivo, 2019-10-23:
Hi.

I've recently been working on migrating a project from a Teensy 3.6 to a Teensy 4. I'm noticing the Teensy 3.6 is much faster updating the screen using DMA. 

It looks like the SPI clock speed setting for the Teensy 4 is not ignored, but capped at a fairly slow speed. I've tried allowing the library to allocate the frame buffer as well as allocating it myself using DMAMEM.

I'm wondering if anyone else is seeing similar speed differences? Any workarounds?

Thanks in advance for the help.

Arduino 1.8.10
Teensyduino 1.48

    Teensy 3.6
    240 Mhz
    
    #define ILI9341_SPICLOCK 30000000
    frameCount() = 24 fps
    
    #define ILI9341_SPICLOCK 60000000
    frameCount() = 45 fps

    Teensy 4.0
    600MHz
    
    #define ILI9341_SPICLOCK 144000000u
    frameCount() = 28 fps
    
    #define ILI9341_SPICLOCK 72000000u
    frameCount() = 28 fps
    
    #define ILI9341_SPICLOCK 36000000u
    frameCount() = 19 fps
    
    #define ILI9341_SPICLOCK 18000000u
    frameCount() = 12 fps
    
    #define ILI9341_SPICLOCK 999999999u
    frameCount() = 28 fps

KurtE, 2019-10-23:
Sorry, I don't have too much time to look into this. 

If you need higher SPI speeds, then maybe you need to change which clock is used to control SPI. 
In particular what is the setting for CCM_CBCMR register. 

I think by default we choose The second clock:
```
CCM_CBCMR = (CCM_CBCMR & ~(CCM_CBCMR_LPSPI_PODF_MASK | CCM_CBCMR_LPSPI_CLK_SEL_MASK)) |
		CCM_CBCMR_LPSPI_PODF(6) | CCM_CBCMR_LPSPI_CLK_SEL(2); // pg 714
```
Actually it is now page 1112... 
Which from our beginTransaction code looks like it starts off with 528MHZ clock going into SPI subsystem.  If you try changing that (2) to (1), or change it after begin is called to (1), then I believe it will feed a 720mhz clock into SPI.... 

Also need to look at the PODF fields of that as well...  

You might ask this types of question on the Forum and maybe will have some time to look again...

mr-stivo, 2019-10-26:
Hi Kurt,

I see there was some action in the PJRC forums regarding the SPI speed. Lucky timing for me.

I'm using your modified SPI library and setting #define ILI9341_SPICLOCK 80000000. frameCount() is around 61-62 fps. Very fast.

Thanks for the help and thank you so much for your libraries.  I'll ask future questions in the PJRC forums.

closed 2019-10-26 by mr-stivo