KurtE/ILI9341_t3n #24: Make Compatible for an ILI9486?

sepp89117, 2020-03-18:
Hello KurtE,

can you help me make the lib compatible for an ILI9486?
I changed the resolution of the display in lib and tested different initializations.
Unfortunately, I fail. My screen stays white.

Please help :)

KurtE, 2020-03-18:
Have you tried our: ILI9488_t3 library?  I believe we have a version of it that is shipping with Teensyduino, and also I believe that the latest changes are up on @mjs513 github.   My version is probalby in sync as well.

sepp89117, 2020-03-18:
Hello, I have now tried it with the graphics test example. I didn't know that 9488 was compatible with 9486.
Unfortunately my screen remains white. The screen may be defective. Teensy works and I measured my wiring. I can't find a mistake. :(
Nevertheless many thanks!

sepp89117, 2020-03-18:
Okay, the display works with an Arduino Mega and the Waveshare library.

I use this display:
https://www.waveshare.com/wiki/4inch_TFT_Touch_Shield#About_Waveshare_ILI9486_Library

Display works with:
https://github.com/MHotchin/Waveshare4InchTftShield
and an Arduino Mega

My PIN setup is:
````#define TFT_DC 9
#define TFT_CS 10
#define TFT_RST 15
#define TFT_MOSI 11
#define TFT_SCLK 13
#define TFT_MISO 12
````
The setup looks like this:
````tft.begin ();
tft.useFrameBuffer (1);
tft.setRotation (3);
````

Any idea why Teensy doesn't work? A problem with chip select?

KurtE, 2020-03-18:
Note: I typically prefer to help out more on forum than on issues like this, as maybe someone has already solved the issue or has a working... 

So again would prefer if you posted up on PJRC maybe in one of the threads like:
https://forum.pjrc.com/threads/55735-ILI9488_t3-Support-for-the-ILI9488-on-T3-x-and-beyond

or  https://forum.pjrc.com/threads/59456-TFT-3-5-quot-display-(320x480)-ILI9488-or-HXD8357D

or ???

sepp89117, 2020-03-18:
I'm sorry. I have already searched for a remedy and found none. I will turn to the forum.

Thank you

closed 2020-03-18 by sepp89117

KurtE, 2020-03-18:
Note: I put a comment up in one of the threads asking if anyone else has one?   The only one I have I believe is parallel interface only.