KurtE/ILI9341_t3n #28: Filename cases do not match

Blackaddr, 2020-04-05:
Hi Kurt! I wanted to begin testing this on Teensy 4.0. I'm guess you might be working in a Windows environment? When using this on Linux, it won't compile because the filename cases in the source file #include's do not match the actual filename cases for the "ILI" prefix.

E.g. In "ili9341_t3n_font_ComicSansMS.c" you have
`#include "ILI9341_t3n_font_ComicSansMS.h"`

but the actual file is called "ili9341_t3n_font_ComicSansMS.h".

I suggest you rename your files to "ILI" captitalized prefix in order to be consistent with most variants of this library.

Blackaddr, 2020-04-05:
It's actually just the one file with the case mismatch, the ComicSansMS.c file. If you want to keep the lowecase "ili" for your custom files you just need to fix the one.

KurtE, 2020-04-05:
I believe that you fixed this with your PR #29 
Thanks again

closed 2020-04-05 by KurtE