KurtE/ILI9341_t3n #33: improper 'len' in

paynterf, 2021-03-03:
It seems the ' len - 2' limit in the for loop is incorrect. When I try to have it print "next", all I get is 'ne'.  In order to print "next", I need to specify "next<sp><sp>"

Am I missing something here?

TIA,

Frank

```
int16_t ILI9341_t3n::drawString1(char string[], int16_t len, int poX, int poY) {
  int16_t sumX = 0;
  uint8_t padding = 1 /*, baseline = 0*/;

  uint16_t cwidth =
      strPixelLen(string); // Find the pixel width of the string in the font
  uint16_t cheight = textsize_y * 6;

  Serial.printf("cwidth = %d, cheight = %d\n", cwidth, cheight);

  if (textdatum || padX) {
    switch (textdatum) {
    case TC_DATUM:
      poX -= cwidth / 2;
      padding += 1;
      break;
    case TR_DATUM:
      poX -= cwidth;
      padding += 2;
      break;
    case ML_DATUM:
      poY -= cheight / 2;
      // padding += 0;
      break;
    case MC_DATUM:
      poX -= cwidth / 2;
      poY -= cheight / 2;
      padding += 1;
      break;
    case MR_DATUM:
      poX -= cwidth;
      poY -= cheight / 2;
      padding += 2;
      break;
    case BL_DATUM:
      poY -= cheight;
      // padding += 0;
      break;
    case BC_DATUM:
      poX -= cwidth / 2;
      poY -= cheight;
      padding += 1;
      break;
    case BR_DATUM:
      poX -= cwidth;
      poY -= cheight;
      padding += 2;
      break;
      /*
   case L_BASELINE:
     poY -= baseline;
     //padding += 0;
     break;
   case C_BASELINE:
     poX -= cwidth/2;
     poY -= baseline;
     //padding += 1;
     break;
   case R_BASELINE:
     poX -= cwidth;
     poY -= baseline;
     padding += 2;
     break;
     */
    }
    // Check coordinates are OK, adjust if not
    if (poX < 0)
      poX = 0;
    if (poX + cwidth > width())
      poX = width() - cwidth;
    if (poY < 0)
      poY = 0;
    // if (poY+cheight-baseline >_height) poY = _height - cheight;
  }
  if (font == NULL) 
  {
    for (uint8_t i = 0; i < len - 2; i++) 
    {
      drawChar((int16_t)(poX + sumX), (int16_t)poY, string[i], textcolor,
               textbgcolor, textsize_x, textsize_y);
      sumX += cwidth / (len - 2) + padding;
    }
  } else {
    setCursor(poX, poY);
    for (uint8_t i = 0; i < len - 2; i++) {
      drawFontChar(string[i]);
      setCursor(cursor_x, cursor_y);
    }
  }
  return sumX;
}
```

KurtE, 2021-03-03:
Will take a look.

@mjs513 - Do you remember anything about this? 

Thanks
Kurt

mjs513, 2021-03-03:
Yes I do - i added it going to look at it now.  I do remember that drawString1 does not get called directly and thought I had tested it.  But will double check.

@paynterf can you post the sketch you are testing with? or just that piece of it.

mjs513, 2021-03-03:
`It seems the ' len - 2' limit in the for loop is incorrect. When I try to have it print "next", all I get is 'ne'. In order to print "next", I need to specify "next"`

Ok think I know what happened.  drawString1 is essentially a helper function for the draw types (number, string, float) and is not meant to be called directly.  You should be calling from your sketch drawString(String, posX, posY). 

The drawString function:
```
  int16_t len = string.length() + 2;
  char buffer[len];
  string.toCharArray(buffer, len);
  return drawString1(buffer, len, poX, poY); 
```
adds 2 to what string length before passing it to drawString1

mjs513, 2021-03-04:
@KurtE 
Just noticed that drawString1 is in the readme for the library.  When you get a chance can you delete it.  

Thanks
Mike

closed 2021-03-04 by KurtE

paynterf, 2021-03-04:
Maybe make the DrawString1 function private?

PR #34 (2021-03-04): fix DrawString 1
Before the only caller was drawString which padded the length passed in to add 2, so drawString1 reduced count by 2...

Instead pass in correct count from drawString...

I think this is the only thing needed to fix #33?

@mjs513 - does this look correct? 

Could just remove it from PDF, but makes sense to have method where I pass in buffers as I always avoid using Strings...

Note: wonder why it was not also just an overload of the drawString method like:
```
int16_t ILI9341_t3n::drawString(const char string[], int16_t len, int poX, int poY) {
```
Also if this is correct, prob [... cut]