KurtE/ILI9341_t3n #35: Text bounds not being reported correctly?

paynterf, 2021-03-04:
I am using the 'getTextBounds' function to retrieve the bounds for text to be placed on top of a filled rectangle, so I can place the text in the middle of the rectangle

However, it seems both the width and height are much larger than the actual number of pixels occupied by the text.  By changing the dimensions of the underlying rectangle, I could approximate the on-screen dimensions of a text string, and the approximate dimension were approximately 1/2  the reported values.

Here's a snippet to demo the problem

 ```
 //inline void CustomBox::Draw(ILI9341_t3n * disp, char* buff, int size)
  String boxstr = "NEXT";
  tft.setTextDatum(MC_DATUM);
  int16_t bounds_x = 0; int16_t bounds_y = 0; uint16_t bounds_w = 0; uint16_t bounds_h = 0;
  tft.getTextBounds(boxstr, NextBox.x, NextBox.y, &bounds_x, &bounds_y, &bounds_w, &bounds_h);

  Serial.printf("bounds are %d,%d,%d,%d\n", bounds_x, bounds_y, bounds_w, bounds_h);
  NextBox.Draw(&tft, boxstr, 3);
```

When this was run, I got the following

`bounds are 160,180,120,40
`

but a filled rectangle with dimensions of 

```
const int NEXTBOX_X = 160;
const int NEXTBOX_Y = 180;
const int NEXTBOX_COLOR = ILI9341_GREEN;
const int NEXTBOX_WIDTH = 70;
const int NEXTBOX_HEIGHT = 20;
 
```
just encloses the text

This is Orientation 1 (0<=x<=320, 0<=y<=240) if it matters

KurtE, 2021-03-11:
Sorry, it has been awhile since I looked at any of this stuff.  I am not sure if @mjs513 remembers any better than I do?

If I remember correctly the code probably uses the full text width and logical height of the text to where the start of the next character would draw and likewise where the start of the next row of text would be.  

And especially with height this can be a relative large distance.

mjs513, 2021-03-11:
Unfortunately I don't remember any better than @KurtE - its been a long while since I looked at it as well.

paynterf, 2021-03-12:
OK, I'll poke around some more & see if I can figure out what is going on

Frank

KurtE, 2021-03-13:
You might try syncing my latest changes and see if that makes a difference.

closed 2021-04-15 by KurtE