KurtE/ILI9341_t3n #40: setscrollmargins

CorBer, 2021-05-14:
The current calculation (thanks for adding this) follows the adafruit example which has a flaw:

uint16_t middle = _height - top + bottom;
The ILI9341 expects 3 numbers following the 0x33 command, those numbers together should equal the displayheight (320pixels). So uint16_t middle=_height-top-bottom;

KurtE, 2021-05-14:
You are right... See what happens when I take the Adafruit stuff without thinking about it ;) 
```
void Adafruit_ILI9341::setScrollMargins(uint16_t top, uint16_t bottom) {
  // TFA+VSA+BFA must equal 320
  if (top + bottom <= ILI9341_TFTHEIGHT) {
    uint16_t middle = ILI9341_TFTHEIGHT - top + bottom;
    uint8_t data[6];
    data[0] = top >> 8;
    data[1] = top & 0xff;
    data[2] = middle >> 8;
    data[3] = middle & 0xff;
    data[4] = bottom >> 8;
    data[5] = bottom & 0xff;
    sendCommand(ILI9341_VSCRDEF, (uint8_t *)data, 6);
  }
}
```

Thanks, pushed up change...
Although I simply did the (top + bottom) just to make sure it was clear

closed 2021-05-14 by KurtE

CorBer, 2021-05-14:
Thanks, yeah that issue is at adafruit also for some time now ...

KurtE, 2021-05-14:
Yep - I pointed it out to LadyAda, who of course said: well issue a PR ;) 
So just issued PR # 72