KurtE/ILI9341_t3n #46: missing API calls for use with GUIslice

u1550l, 2023-03-12:
Hello,
while trying to use SPI1 instead of SPI, I encountered:
GUIslice requires class ILI9341_t3n member functions measureTextWidth() and measureTextHeight() 

PaulStoffregen added them to ILI9341_t3 with commit https://github.com/PaulStoffregen/ILI9341_t3/commit/4c4558fd31dd5266e139096687e7b36ddeabe469
"Add string measurement APIs"

Could you please help me to merge these functions?

KurtE, 2023-03-12:
Probably could add these, however, it would still require changes to GUISlice to include <ILI9341_t3n.h>

And if doing that, we do have the same text measuring stuff built in as some of the other drivers like:
RA8875 or 76 and it looks like the default measure function in guiSlice.
```
  void getTextBounds(const uint8_t *buffer, uint16_t len, int16_t x, int16_t y,
                     int16_t *x1, int16_t *y1, uint16_t *w, uint16_t *h);
  void getTextBounds(const char *string, int16_t x, int16_t y, int16_t *x1,
                     int16_t *y1, uint16_t *w, uint16_t *h);
  void getTextBounds(const String &str, int16_t x, int16_t y, int16_t *x1,
                     int16_t *y1, uint16_t *w, uint16_t *h);
```
And the <Ili9341_t3.h> version does not handle the case where the text may draw, not starting at the X, Y
passed in.  which is why GUISlice hs a comment todo support for baseline info...

And the implementation, that I would probably do is something like:
```
uint16_t ILI9341_t3::measureTextWidth(const char* text, int n) {
	int16_t x1, y1;
        uint16_t w, h;
	if (n == 0)  n = strlen(text);
	getTextBounds(text, n, 0, 0, &x1, &y1, &w, &h);
	return w;
}
```
The height one would be identical, except return h instead of w.

u1550l, 2023-03-12:
In fact, I came across these missing functions when I already started to work on the GUIslice adapting code.
I think, I can live with the "todo"-limitation.
I will try your suggested code tomorrow.

u1550l, 2023-03-13:
I'd like to give you a success feed back. Thanks a lot.
The display was on within some minutes.

PR #47 (2023-03-13): Add measureTextWidth/Height
resolves: #46

Added in the two methods for more compatibility with ili9341_t3 driver.  Note: These are simply calling the method getTextBounds, which has additional functionality.